HEN3DRIK/BeadSorter #3: Probleme mit Servo in Bauelement 3

Fapi097, 2023-03-01:
Vorweg - beim Bau der Sorting machine, gab es weder elektrisch, noch drucktechnisch Probleme, außer bei Bauelement **"3-analyzer"**. Genauer gesagt liegt das Problem beim Servomotor, auf welchem **"3-analyzer-separation-arm"** befestigt ist und mit **"3-analyzer-exit-base"** sowie **"3-analyzer-exit-tube"** verbunden ist. Die Bügelperlen werden in diesem Abschnitt nicht richtig gefördert - das Problem liegt sowohl in der Aufnahme als auch Abgabe. Die Perlen scheinen sich im Übergang zwischen **seperation arm** und **analyzer exit tube** zu verfangen, obwohl ich alle mir erdenklichen Befestigungsmöglichkeiten und Servostellungen augeschöpft habe. (Zur besseren Darstellung habe ich eine Bilderstrecke sowie Videos über einen Dropboxlink zusammengefasst.) [https://www.dropbox.com/scl/fo/sogxwpxyg6pxwu0e42v44/h?dl=0&rlkey=3x28gm473k9n9inqbctwqn96i]
In der Software erschien mir die Möglichkeit der Änderung des Wertes in der Zeile _"#define angle 15 //Servo"_ nachvollziehbar, erzielte jedoch keine Wirkung. 
Ich hoffe, dass Sie einen Lösungsweg für mein Problem haben, da ich sehr viel Spaß und Freude an diesem Projekt habe und es auch sehr gerne abschließen möchte.
Ich wünsche noch einen angenehmen Abend.
[image]

Mit freundlichen Grüßen

Fabian

HEN3DRIK, 2023-03-02:
Hi,

_Angle_ ist für die initiale Position des Arms (wird nur in der setup routine des code verwendet). Dieser muss so gewählt werden, dass er zu beginn bündig unter dem entry tube steht. Danach wird die funktion serveFeedOut() verwendet um die Bügelperle zum exit tube zu befördern. die werte 41,44,41,44,... verursachen ein wackeln, dass die perle hoffentlich herausbefördert. beim erneuten durchgang wird servoFeedIn() verwendet um unter dem entry tube ebenfalls etwas zu wackeln.

wie du sicherlich bemerkt hast dies der kniffeligste teil des sortierers. hier muss alles sehr glatt sein. es sind teilweise kleinste einstellungen, die dazuführen, dass die perle sich verheddert. ich erinnere mich wie lange ich an diesem mechanismus saß. 

was du machen kannst:
- alles glatt schmirgeln, alle röhren, alle flächen. evtl. mit etwas PTFE spray abputze. auf deinen fotos sieht es so aus als hättest du das nicht gemacht. die röhren von innen richtig schön glatt feilen. zumindest die kurzen direkt am analyzer.
- die werte in servoFeedIn und servoFeedOut anpassen. deinen videos nach passen die nicht. du möchtest erreichen, dass der servo genau unter/über den röhren "wackelt".
-  ich hatte die teile auch erst grau gedruckt, danach habe ich alles im analyzer weiß gedruckt, was zu besserer erkennung führte. 

ich weiß, ist richtig nervig. als ich es dann richtig eingestellt hatte lief es aber gut und macht seither keine probleme. 

void servoFeedIn() {
  delay(200);
  servo.write(16);
  delay(200);
  servo.write(18);
  delay(200);
  servo.write(17);
  delay(200);
}

void servoFeedOut() {
  servo.write(41);
  delay(200);
  servo.write(44);
  delay(200);
  servo.write(41);
  delay(200);
  servo.write(44);
  delay(200);
  servo.write(41);
  delay(200);
  servo.write(44);
  delay(200);
  servo.write(41);
  delay(200);
  servo.write(44);
  delay(500);
}

Fapi097, 2023-03-02:
Danke das hilft mir schon sehr ich werde es probieren und mich melden wenn ich es geschaft habe. 
Einen schönen Abend noch mit freundlichen grüße Fabian Schragner

Fapi097, 2023-03-03:
Danke für die hilfe mit dem Servo. Dieser Vorgang funktioniert einwandfrei. Leider hat sich ein anders Problem aufgetan Wenn ich die Maschine starte bewegt sich der stepper motor nicht und auch der farb scan auch nicht der servo bewegt sich 4 mal dannach stellt er seinen arbeit wieder ein nur mehr der  Hopper motor dreht das wars. Ich bin mir nicht ganz sicher ob der Stepper richtig am Driver verbunden ist .Und wieso die  maschine nicht automatisch zum sortieren anfang. vielen dank für ihre hilfe und ein Schönes Wochende noch mit freundlichen grüße Fabian

Daffy2905, 2023-03-09:
Bin auch gerade in der finalen Phase knapp vor Inbetriebnahme :-) Ich hatte ein änliches Problem mit der Analyzer und das die Perlen sich manchmal zwischen EntryTeil und Arm "verkeilen". Hab mich dann etwas mit dem Code vertraut gemacht und die von Hendrik erwähnten Werte adaptiert. Dies führte zu einer merklichen Verbesserung. Wie auch Hendrik geschrieben hat, alles muss glatt sein wie ein Babypopo. Die Toleranzen bewegen sich da im 0,irgendwas mm Bereich. Aber grundsätzlich bin ich absolut begeistert von dem Teil :-)

Fapi097, 2023-03-09:
ja Daffy2905 es ist ein wirklich cooles Projekt darf ich dir paar fragen stellen 1.wie hast du den Steppermotor angeschlossen 2. bei mir sortiert die Maschine nicht im autostart und wie bekomme ich die farben konfiguriert. besten dank und einen schönen abend noch

Daffy2905, 2023-03-09:
1. Wie am Schema dargstellt, funktioniert soweit
2. Ich würde mal ein USB-Kabel an Arduino während dem Betrieb anschließen und mit dem Serial Monitor der Arduino Software mitschauen, was das Teil macht. Da bekommst im Log mit, was nicht passt .....
LG

Fapi097, 2023-03-12:
im log steht zwar color reading aber es passiert nichts haha

Fapi097, 2023-03-12:
soll ich mal den Log posten

Fapi097, 2023-03-12:
[image]

Daffy2905, 2023-03-13:
Passiert überhaupt nichts? Also beim Start sollte der ServoMotor im Analyzer 6 Mal hin- und her fahren, es wird die initiale Farbkalibrierung durchgeführt. Anschließend sollte der HopperMotor starten und der Spaß beginnt. Spannungen alle überprüft? Die Verkabelung ist im ersten Moment etwas tricky, aber nach längerer Beschäftigung wird vieles schnell klar ...

Fapi097, 2023-03-13:
ja das macht er auch aber dannach passiert nichts mehr also er fährt 6 mal hin und her de hopper motor fördert aber das wars der servo arm bleid dann einfach stehen mit einer Bügelperle drinnen und das wars

HEN3DRIK, 2023-03-14:
Es liegt vermutlich an dieser Stelle in Zeile 123..125

```
if (digitalRead(setupPin) == HIGH) {
  while (digitalRead(setupPin) == HIGH) {
  }
```

Wenn der Taster gedrückt ist schaltet das Gerät in den manuellen Modus. Überprüf mal die Verkabelung oder kommentier diese Bedingung testweise aus (123-131)

Fapi097, 2023-03-14:
Oky ich probiere es mal Danke schön Tag noch

Fapi097, 2023-03-15:
Es tut mir wirklich leid das ich so nerv tötend bin mit meinen Fragen. Ich habe die Verkabelung anders umgesetzt und ich weiß nicht ob das zu 100 % richtig ist.

Fapi097, 2023-03-15:
[image]
Ist der Schaltplan so richtig.

Daffy2905, 2023-03-15:
Wieviel Volt gibt der Nano dann aufs BreakoutBoard aus? Wennst nur mit 5V in den Nano reingehst kommen am V+-Ausgang ned mehr als 4V raus für die anderen Komponenten wie der FarbSensor ....Von meinem/meiner Galaxy gesendet

Daffy2905, 2023-03-15:
Und Dir fehlen beim StepperModul die 12V ...Von meinem/meiner Galaxy gesendet

Fapi097, 2023-03-16:
Danke für die hilfe ich werde das mal überprüfen

Fapi097, 2023-03-19:
habe alles nochmal nachgebaut jetzt sortiert er Automatisch aber der knopf funkoniert nicht also das mit die vier farben wie gehört der taster richtig angeschlossen . Aber er sagt immer color not found auser 2 farben gelb und blau 
[image]

Daffy2905, 2023-03-19:
Ich kämpfe ständig mit der Zuverlässigkeit des Förderungsprozesses. Mal bleiben die Perlen beim Eingang Analyzer hängen, dann verkeilen sie sich zwischen Arm und Eingang Analyzer oder bleiben unten im Verteilerarm drinnen hängen. Hab schon alles geschliffen, PTFE-Spray drauf, aber so schön flüssig wie im Video von HENDRIK läufts leider bei mir - und leider steigt auch schon langsam der Frustrationsgrad, obwohl ich vom Projekt echt echt begeistert bin .....

Fapi097, 2023-03-19:
Ja Kenn ich gut ich habe das gelöst mit den einstellen im adruino code  mit denn werten einstellen das funktioniert super
Aber ich hab mit der Dispatcher problem also das er richtig dreht und die farbe erkennt ich verstehe das nicht ganz wie das geht bei mir steht nur immer new container haha ja bei mir  steigt leider auch immer mehr die Frustration aber vieleicht können wir uns gegenseitig helfen.

HEN3DRIK, 2023-03-19:
Ich kann euch gut verstehen. Bei dem Projekt braucht man starke Nerven. Auch ich startete sehr euphorisch und am ende habe ich dann die software neugeschrieben und zig Teile verändert, dass es überhaupt mal lief. Was ich sagen will, es zahlt sich mehr oder weniger aus. Wenn das Ding dann mal 2000 Perlen durchsortiert hat freut man sich schon sehr. In der ganzen Zeit hätte man es aber auf per Hand sortieren können, aber das ist ja nicht die Motivation 😅. Ich habe ihn letzten auch noch einmal angeworfen um nachzugucken wie etwas genau funktioniere und dann habe ich ihn mal laufen lassen lassen: klappt immer noch gut. Sorry, dass ich so selten dazu komme zu antworten:

Zu den Problemen: 
@Daffy2905 ich hatte oft deformierte Perler drin, die verstopfen alles. Ansonsten macht es sinn das Röhrchen über dem Analyzer und das exit Röhrchen mit gröberen schleifpapier in eine richtung zu weiten (wie so rillen = wenig kontaktfläche) und dann ptfe dran. aber das ptfe sparsam und "polieren". Im Verteilerarm: Wo genau? bei mir manchmal in den organischen Röhren. Keine Ahnung wie sich das verkeilt, ist wohl alles eine Frage den Anzahl an Perlen.

@Fapi097 Der Sorter hat eine Kapazität von 11 Plätzen für Farben und eine Learning/NichtErkannt Platz. Wenn der Sorter im AutoSort ist (Knopf nicht gedrückt) lernt er die Farbe, wenn er sie nicht kennt an und wirft sie sofort wieder aus (not found + not allContainerFull -> StoreColor). Wir diese Farbe nun das NÄCHSTE mal erkannt, fährt der dispatcher auf die vorher gespeicherte Position. Es ist wichtig zu verstehen, dass der Sorter die Farben lernt/speichert (max 16 stück in tempStoredColors) und diese in dynamicContainerArray (max 11) zuordnet (je nachdem welche zuerst kommt).

Viel Glück und Durchhaltekraft!

Fapi097, 2023-03-20:
Danke das hat mir sehr viel geholfen also einfach mal sortieren lassen eine zeitlang habe den Vorgang immer abgebrochen weil ich die Meldung in Serial Monitor gesehn haben. Und für was ist dann der Modus mit dem Knopf.
Ich werde das Projekt sicher bis zum Schluss durchhalten ich danke dir wirklich für deine Hilfe.

Fapi097, 2023-03-24:
Guten Morgen @Daffy2905 bei mir Funktioniert jetzt alles fast reibungslos nur die Drehscheibe vom Hopper ist ein Problem weil die Perlen nicht durch fallen und sich verkeilen zwischen Scheibe und Hopperauslass. Welchen Motor benutzt und wie schnell dreht der.

Daffy2905, 2023-03-24:
Freut mich - also die Geschwindigkeit des Hoppermotors (hab den DC6V 30RPM) kannst Du entweder über die Spannung regeln, hab glaub ich bei mir 7V als auch über die Zeile 11 (#define motorSpeed 255 //Container Motor) - stell hier mal einen Wert von unter 100 ein, da solltest schon eine Drehgeschwindigkeitsreduktion sehen .....

Fapi097, 2023-03-24:
@Daffy2905 ja das hab ich schon alles ausgetestet aber die Perlen werden immer eingeklemmt zwischen Drehscheibe und Dem Loch habe aber alles geschliffen und ausgerichtet aber die Bleiben immer stecken.

Fapi097, 2023-03-24:
@Daffy2905 habe auf 150 eingestellt hat aber nichts gebracht es kommt immer so raus wie auf den foto das es sich verkeilt.
[image]
[image]