HEN3DRIK/BeadSorter #6: Attaching tubes to plate

konforce, 2023-10-29:
[image]

I've got most of the project assembled, but am not understanding how the tubes are supposed to attach. Is the intention that the tabs of the tubes are supposed to fit in the corresponding slots in the rings?

On the plates.stl file, only three of the rings have slot cut out. (On the attached image, it is the three nearest ones.) Is this an incomplete STL or am I misunderstanding?

Also, did you glue the tubes or just friction fit?

HEN3DRIK, 2023-10-29:
Hi, i think i tried adding some alignement slots, but it's just super glueing them onto the ring. 😉

konforce, 2023-10-30:
There's three different versions of the slides - is the intention to print out four of each?

HEN3DRIK, 2024-08-16:
yes