HEN3DRIK/BeadSorter #9: Micro stepping configuration

iliar-rabet, 2024-08-16:
Hello,

Thanks for sharing this. I have set up everything but the machine is not very accurate. On the A4988, I have used the following set up for micro stepping:
MS1 : H
MS2: H
MS3: L
This would give me a step resolution of 1/8 of the full. Is this what you have used?

Second question: some times the stepper motor stops at some locations that are blocked and there is no container. 
Do we have to manually spin the tube so it is initialized at some specific point?

Bests,
Iliar

HEN3DRIK, 2024-08-16:
Hi,

i connected to MS2 + MS1.
then I set #define stepperMulti 100 //Stepper

This gave me a very fine resolution.
[image]

The stepper needs to be in a certain position 
[image]

iliar-rabet, 2024-08-16:
OK, thanks so much.
It is kind of hard to be sure which tube is the initial position by looking at this picture. 
Is it tube number 5 if we count anti-clockwise starting from the stepper?

HEN3DRIK, 2024-08-16:
It actually doesent matter. The stepper moves clockwise and ist okay to start on the 3rd tube of each quarter. the 2nd (in this case the left next to picture position) is the last tube of the array (12th) so it will move there first after scanning a new bead. When the scanned bead is recognized after that it'll be sorted in 1 (position on the picture) 2,3 and so on. It wil skip the beam between pos 1 and 2 automatically.